jpnurmi/flutter_libserialport #130: After creating a new Flutter project, following the steps of Pubdev to add dependencies and sample code, an exception occurred after running

xigexb, 2025-03-12:
After creating a new Flutter project, following the steps of Pubdev to add dependencies and sample code, an exception occurred after running

Flutter assets will be downloaded from https://storage.flutter-io.cn. Make sure you trust this source!
Launching lib\main.dart on M2007J22C in debug mode...
Running Gradle task 'assembleDebug'...
Your project is configured with Android NDK 26.3.11579264, but the following plugin(s) depend on a different Android NDK version:
- flutter_libserialport requires Android NDK 27.0.12077973
Fix this issue by using the highest Android NDK version (they are backward compatible).
Add the following to E:\ucthings\rfid_fibre_tool\android\app\build.gradle.kts:

    android {
        ndkVersion = "27.0.12077973"
        ...
    }

FAILURE: Build failed with an exception.

* What went wrong:
A problem occurred configuring project ':flutter_libserialport'.
> com.android.builder.errors.EvalIssueException: [CXX1110] Platform version 16 is unsupported by this NDK. Please change minSdk to at least 21 to avoid undefined behavior. To suppress this error, add android.ndk.suppressMinSdkVersionError=21 to the project's gradle.properties or set android.experimentalProperties["android.ndk.suppressMinSdkVersionError"]=21 in the Gradle build file.

* Try:
> Run with --stacktrace option to get the stack trace.
> Run with --info or --debug option to get more log output.
> Run with --scan to get full insights.
> Get more help at https://help.gradle.org.

BUILD FAILED in 1s
Error: Gradle task assembleDebug failed with exit code 1

xigexb, 2025-03-12:
When I follow the relevant document settings, the following error will appear

Launching lib\main.dart on M2007J22C in debug mode...
Running Gradle task 'assembleDebug'...
e: file:///C:/Users/abc/AppData/Local/Pub/Cache/hosted/pub.flutter-io.cn/flutter_libserialport-0.5.0/android/src/main/kotlin/org/sigrok/flutter_libserialport/FlutterLibserialportPlugin.kt:10:48 Unresolved reference: Registrar

FAILURE: Build failed with an exception.

* What went wrong:
Execution failed for task ':flutter_libserialport:compileDebugKotlin'.
> A failure occurred while executing org.jetbrains.kotlin.compilerRunner.GradleCompilerRunnerWithWorkers$GradleKotlinCompilerWorkAction
   > Compilation error. See log for more details

* Try:
> Run with --stacktrace option to get the stack trace.
> Run with --info or --debug option to get more log output.
> Run with --scan to get full insights.
> Get more help at https://help.gradle.org.

BUILD FAILED in 16s
Error: Gradle task assembleDebug failed with exit code 1

saeedvw, 2025-03-19:
i have the same issue and even when i create a new flutter project as soon as i add flutter_libserialport it fails to build.

seanyweany, 2025-03-27:
Same exact thing happened here! Forced the correct SDK and NDK then boom, "Registrar" exception...

saeedvw, 2025-03-27:
> Same exact thing happened here! Forced the correct SDK and NDK then boom, "Registrar" exception...

A temporary solution is to comment the line where it imports Registrar in the kotlin file
```
import io.flutter.plugin.common.PluginRegistry.Registrar
```
It will work, the problem is that this plugin uses the old plugin api. The real solution is to migrate this library to the new Plugin api.

loafman-kangjun, 2025-04-26:
l have the same issues, but how to solve it?

ImadMAKS, 2025-05-14:
> This is solved by [#129](https://github.com/jpnurmi/flutter_libserialport/pull/129). Current fix by adding to `pubspec.yml`.
> 
> ```
> # Temporary fixes
> dependency_overrides:
>   # because of https://github.com/jpnurmi/flutter_libserialport/issues/128
>   flutter_libserialport:
>     git:
>       url: https://github.com/kapil708/flutter_libserialport
>       ref: flutter_3.29.0
> ```

This workaround worked in my case, thanks!